pouriap/TinyRF #1: TinyRF on ESP-12F

lpriemis, 2021-06-10:
High not sure if this is the place to put it. But i'm trying to build a sensor with Attiny13 which sends it's values over 433MHZ to a Bridge which will put it on MQTT. The transmitter part works perfectly i've tested it using a Arduino Nano as a receiver. But when i try to upload the receiver code to an ESP-12F it keeps on rebooting and gives me the following feedback:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

I have the feeling it has something todo with the interrupts or something. Can somebody please share if it should be working with an ESP-12F and if so what can i be doing wrong?

pouriap, 2021-06-11:
@lpriemis You need to add `IRAM_ATTR ` before ISR functions for the ESP. 

To do that you should edit `TinyRF_RX.cpp` and add that to [line 211](https://github.com/pouriap/TinyRF/blob/83fa498b37ca274775bb70593623d1bca7bd1094/src/TinyRF_RX.cpp#L211) as follows:
```IRAM_ATTR  void interrupt_routine(){...```

If that didn't work try using another pin for RX, for example if using pin 4:
```void setupReceiver(4);```

This library was written with only Arduino as receiver in mind but I know at least one person who has used it with ESP without issues. I plan to add support for ESP in the next release so people won't have to do this edit manually.

lpriemis, 2021-06-11:
@pouriap You are right. It worked. 

Unfortunatly now i get the reboots as soon as i want to connect to WIFI ("ESP8266WiFi.h").

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40100728: umm_malloc_core(size_t) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 433
EXCVADDR: 0x4006eca1

Decoding stack results
0x40100910: malloc(size_t) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x40100b60: calloc(size_t, size_t) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 902
0x4010028c: pvPortZalloc(size_t, char const*, int) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\heap.cpp line 301
0x40100c3d: __wrap_spi_flash_read(uint32_t, uint32_t*, size_t) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_phy.cpp line 309
0x40100657: umm_free_core(void*) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40100c3d: __wrap_spi_flash_read(uint32_t, uint32_t*, size_t) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_phy.cpp line 309
0x40201e6e: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 189
0x40201dc3: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 157
0x40100900: malloc(size_t) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 529
0x40205191: uart_flush(uart_t*) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\uart.cpp line 544
0x402050f9: uart_write(uart_t*, char const*, size_t) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\uart.cpp line 509
0x402050f9: uart_write(uart_t*, char const*, size_t) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\uart.cpp line 509
0x40205665: uart_set_debug(int) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\uart.cpp line 933
0x402037b8: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x40201f00: ESP8266WiFiSTAClass::begin(char*, char*, int, unsigned char const*, bool) at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 216
0x402010f8: setup_wifi() at C:\Users\lpriemis\OneDrive - A.S. Watson Europe\Documents\Arduino\433_TinyRF_RX_ESP12/433_TinyRF_RX_ESP12.ino line 127
0x4020115e: setup() at C:\Users\lpriemis\OneDrive - A.S. Watson Europe\Documents\Arduino\433_TinyRF_RX_ESP12/433_TinyRF_RX_ESP12.ino line 66
0x40204350: loop_wrapper() at C:\Users\lpriemis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 194

pouriap, 2021-06-12:
Does the example sketch work?

lpriemis, 2021-06-12:
Yes it works like a charm right up till the point i try to connect to WIFI (  WiFi.begin(ssid, password); ). Then it start rebooting: 

```
#include "TinyRF_RX.h"
#include "ESP8266WiFi.h"

// you can only use pins that support external interrupts
// in Arduino Uno this is pins 2 and 3
int rxPin = 4;

char ssid[] = "XXXXXXXXXX";
char password[] = "XXXXXXXXXXX";

void setup(){
  Serial.begin(115200);
  setupReceiver(rxPin);
  WiFi.begin(ssid, password);
}

void loop(){

  const uint8_t bufSize = 30;
  byte buf[bufSize];
  uint8_t numLostMsgs = 0;
  uint8_t numRcvdBytes = 0;

  // number of received bytes will be put in numRcvdBytes
  // if sequence numbering is enabled the number of lost messages will be put in numLostMsgs
  // if you have disabled sequence numbering or don't need number of lost messages you can omit this argument
  uint8_t err = getReceivedData(buf, bufSize, numRcvdBytes, numLostMsgs);

  // the receiver has a circular FIFO buffer
  // if getReceivedData() isn't called frequently enough then older messages will get overwritten
  // so make sure the frequency at which you send messages in your tx code is slower than the frequency
  // at which you call getReceivedData() in your rx code to prevent that
  // specially when you are using sendMulti()
  // duplicate messages that are sent using sendMulti() will stay in the buffer until you read the first one
  // you can change the buffer size in settings.h

  if(err == TRF_ERR_NO_DATA){
    return;
  }

  if(err == TRF_ERR_BUFFER_OVERFLOW){
    Serial.println("Buffer too small for received data!");
    return;
  }

  if(err == TRF_ERR_CORRUPTED){
    Serial.println("Received corrupted data.");
    return;
  }

  // if sequence numbering is enabled and you use the sendMulti() function for sending a message
  // multiple times then getReceivedData() will return TRF_ERR_SUCCESS only once
  // all the duplicate messages will be automatically ignored
  // this means all you need to do is check if the return code is TRF_ERR_SUCCESS
  // these are non-repeated, crc-valid messages
  if(err == TRF_ERR_SUCCESS){
    Serial.print("Received: ");
    for(int i=0; i<numRcvdBytes; i++){
      Serial.print((char)buf[i]);
    }
    Serial.println("");

    if(numLostMsgs>0){
      Serial.print(numLostMsgs);
      Serial.println(" messages were lost before this message.");
    }
  }
  
}
```

pouriap, 2021-06-12:
A few things to try:
- Make 100% sure that the ESP has enough power, most ESP issues are power related. Try another power source, different wires, etc.
- Wait until WiFi is connected: 
```
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED){
  delay (500);
  Serial.print (".");
}
```
- Go to `Tools -> Erase Flash` and select `Sketch + WiFi Settings` and change your setup code like this:
```
void setup(){
  Serial.begin(115200);
  setupReceiver(rxPin);
  WiFi.mode(WIFI_STA);
  WiFi.setAutoReconnect(true);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED){
    delay (500);
    Serial.print (".");
  }
}
```
- Try changing the baudrate of `Serial`
- Try connecting to WiFi first before calling `setupReceiver()` like so:
```
void setup(){
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  setupReceiver(rxPin);
}
```
- Try another pin for RX
- Try using another version of ESP8266 core

lpriemis, 2021-06-12:
After updating the Library. It seems to work. Now it crashes on MQTT. 

I'm going to have a look with a different MQTT client.

lpriemis, 2021-06-14:
Great news,

I got it to work with ArduinoMqttClient.

So i guess updating ESP8266WiFi and changing to ArduinoMqttClient. Did the trick.

Now i have a sensor with a Attiny13 which cusomes about 10uA in sleep mode and about 1mA during the sensing part (25ms). I guess this can keep running for ever on a 18650 cell. ESP works like a bridge to put the received message on a generic MQTT broker so my Home Assistant can pick it up. Now i'm set to build any type of sensor i want in my house with a Attiny and a 433Mhz transmitor.

Just to be complete, this are my arduino settings for uploading the sketch to the ESP12F:
[image]

closed 2021-06-14 by lpriemis

pouriap, 2021-06-14:
That's great.
Glad I was able to help.