pouriap/TinyRF #2: Compiling error!

HemdrupDk, 2021-07-25:
I got an compiling error:
Arduino: 1.8.15 (Windows Store 1.8.49.0) (Windows 10), Board: "Digispark Pro (Default 16 Mhz)"
....\Documents\Arduino\libraries\TinyRF\src\TinyRF_RX.cpp: In function 'void setupReceiver(uint8_t)':
....\Documents\Arduino\libraries\TinyRF\src\TinyRF_RX.cpp:98:45: error: 'digitalPinToInterrupt' was not declared in this scope

  attachInterrupt(digitalPinToInterrupt(rxPin), interrupt_routine, FALLING);

pouriap, 2021-07-26:
I just released a new update that should fix the issue.

I don't have the Digispark board to test with it, but the sketch compiles without any errors now.

pouriap, 2021-07-31:
@HemdrupDk  Did you try the new release?

closed 2021-09-02 by pouriap