pouriap/TinyRF #3: Question - Attiny13 RX

gitgaatachal, 2021-09-29:
Does the receiver example code work on Attiny13 MCU?

pouriap, 2021-09-29:
Unfortunately not.

It needs a relatively big amount of RAM and Flash memory for its buffer functions. 

I've been meaning to try and write a version that works on ATtiny but I'm too busy for the time being. I'm also not so sure if its even possible.

gitgaatachal, 2021-09-29:
> I've

Oh ok, even I had a doubt after looking into the code and the buffers used. Shall I open a new ticket for RX support for Attiny13? so that when you go over it you can mark the ticket as resolved.

pouriap, 2021-09-29:
That's a good idea.

Then people who are also interested can also subscribe and be notified.

gitgaatachal, 2021-09-30:
Hi,
 
I have opened a new ticket

closed 2021-09-30 by gitgaatachal