pouriap/TinyRF #5: Program does not receive messages

MicheQ, 2021-12-01:
//the program works normally I should get a value of 1,2 or 0 from ledStatse, but if I uncomment the part of the program in void //loop() ledStates is always 0, like I don't get any messages.

//In a simple way the commented part together with the colorWipe () function turns on the LEDs of a neopixel in sequence this //function uses the millis () to have a non-blocking program.
//I don't understand where I'm wrong and if there is any value to change in setting.h .

#include <Adafruit_NeoPixel.h>
#include "TinyRF_RX.h"
#define NUM_PIXELS 16

unsigned long interval = 80;  // the time we need to wait
unsigned long previousMillis=0;

uint32_t currentColor;// current Color in case we need it
uint16_t currentPixel = 0;// what pixel are we operating on

uint8_t rxPin = 2;

char ledStatesChar[2];
byte ledStates = B00;
byte ledState = B00;
byte ledState2 = B00;

Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUM_PIXELS, 6, NEO_GRB + NEO_KHZ800);

void setup() {

  Serial.begin(115200);

  setupReceiver(rxPin);
  currentColor = strip.Color(100,0,0);
  currentPixel = 0;
  strip.begin();
  strip.show(); // Initialize all pixels to 'off'
}

void loop(){

const uint8_t bufSize = 30;
  byte buf[bufSize];
  uint8_t numLostMsgs = 0;
  uint8_t numRcvdBytes = 0;

  uint8_t err = getReceivedData(buf, bufSize, numRcvdBytes, numLostMsgs);

 
  if(err == TRF_ERR_SUCCESS){
    digitalWrite(13, true); // Flash a light to show received good message  
    Serial.print("Received: ");
    for(int i=0; i<numRcvdBytes; i++){
      Serial.print((char)buf[i]);
      ledStatesChar[i] = ((char)buf[i]);  //save the data
      //Serial.println(ledStates);
    }
    Serial.println("");
    
        ledStatesChar[numRcvdBytes] = '\0';
        ledStates = atoi(ledStatesChar);
        digitalWrite(13, false);
        Serial.println(ledStates);
    }

Serial.println(ledStates); 

//if (ledStates == 1){
//  if ((unsigned long)(millis() - previousMillis) >= interval) {
//    previousMillis = millis();
//    colorWipe();
//   }
//}
//else {
//  strip.clear();
//  strip.show();
//  currentPixel = 0;
//  }
//
//if(currentPixel == 15){
// strip.clear();
// strip.show();
//}
//
//}

void colorWipe(){
  strip.setPixelColor(currentPixel,currentColor);
  strip.show();
  currentPixel++;
  if(currentPixel == NUM_PIXELS){
    currentPixel = 0;

  }
}

pouriap, 2021-12-01:
I assume the neopixel library is spending too much time in interrupts and preventing tinryRF from receiving the data correctly.

If using the Adafruit Neopixel is not important can you test with the FastLED library and see if that works. I myself had issues with the Neopixel library with a ESP8266 project and switched to using FastLED.

Also what board are you using as the receiver?

MicheQ, 2021-12-02:
I have notice that: 
else {
  strip.clear();
  strip.show();
  currentPixel = 0;
  }
if unccoment all the rest of code and comment this part the TinyRF works good and receive message, but when ledStates==1 enter in function colorWipe and rest 1 also if send another number. The problem is the strip.show() which perhaps disturbs the serial buffer because until it encounters strip.show() the code works perfectly.
P.S.
I have tryed also with FastLED.

pouriap, 2021-12-02:
Can you post your complete transmitter and receiver sketch?

Also in the meantime, have you tried running the code without the LED strip attached? If it works without the strip attached it may be a power issue and adding a big capacitor to the receiver module could solve it.

MicheQ, 2021-12-02:
Yes I tried as you told me to disconnect the neopixel but nothing changes.
For now the whole receiver code is the one posted. Instead this is a part of transmitter code (although I don't think it's a problem with the TinyRF but with the strip.show () command) :
```
#include "TinyRF_TX.h"
#define intervallo 500

byte ledStateL = B00;
byte ledStateR = B00;
byte ledStates = B00;
byte sendState = 0;
int counterL = 0;
int counterR = 0;
const int buttonLeft = 3;    // the number of the pushbutton pin
const int buttonRight = 4;
int buttonStateL;             // the current reading from the input pin
long lastDebounceTime = 0;  // the last time the output pin was toggled
long debounceDelay = 100;    // the debounce time; increase if the output flickers
int buttonStateR;             // the current reading from the input pin
long lastDebounceTime2 = 0;  // the last time the output pin was toggled
long previusMillis = 0;
char ledStatesChar[2];    //char array; stores the Led States; needed in vw_send function

void setup() {
  // transmitter default pin is pin #2. You can change it by editing Settings.h
  setupTransmitter();
  Serial.begin(115200);

  pinMode(ledL, OUTPUT);
  pinMode(ledR, OUTPUT);
  pinMode(buttonLeft, INPUT);
  pinMode(buttonRight, INPUT);
  
}

void loop() {

  if (sendState == 1) {  //data is only send if a button was pressed
    ledStates = (ledStateL | ledStateR);
    digitalWrite(13, true); // Flash a light to show transmitting
    itoa(ledStates, ledStatesChar, 10);
    sendMulti((char *)ledStatesChar, strlen(ledStatesChar),3);
    delayMicroseconds(TX_DELAY_MICROS);
    digitalWrite(13, false);
  }
  sendState = 0;

  //debounce code button Left:
  int buttonValueLeft = digitalRead(buttonLeft);
  if ((millis() - lastDebounceTime) > debounceDelay) {
    if (buttonStateL != buttonValueLeft) {
      if (buttonValueLeft == HIGH) {
        counterL++;
        if ((counterL % 2 == 0)) {
          ledStateL = B00; //00 = first LED off, 01 = first LED on
        }
        else {
          ledStateL = B01;
          ledStateR = B00;
          counterR = 0;
        }
        sendState = 1;
      }
      lastDebounceTime = millis();
      buttonStateL = buttonValueLeft;
    }
  }

  //debounce code button Right:
  int buttonValueRight = digitalRead(buttonRight);
  if ((millis() - lastDebounceTime2) > debounceDelay) {
    if (buttonStateR != buttonValueRight) {
      if (buttonValueRight == HIGH) {
        counterR++;
        if ((counterR % 2 == 0)) {
          ledStateR = B00;    //00 = second LED off, 10 = second LED on
        }
        else {
          ledStateL = B00;
          ledStateR = B10;
          counterL = 0;
        }
        sendState = 1;
      }
      lastDebounceTime2 = millis();
      buttonStateR = buttonValueRight;
    }
  }
  //Serial.println(ledStateR);

  
}
```

MicheQ, 2021-12-03:
Hi, in the end I solved it in this way and everything works, I also inserted a capacitor near the receiver because if I go up in brightness I also have problems.

Now the function strip.show() is not called every time but only if i receive number 0, and this is fine for my application.

```
void loop() {
.
.
.
 if(err == TRF_ERR_SUCCESS){
     .
     .
     .
     // only if I get the number 0, update the neopixel with all pixels off
     if (ledStates == 0 ){  
          currentPixel = 0;    //counter set to zero for the next iteration
          strip.clear();
          strip.show();
      }      
  }
}
```
Tanks for the support and the quickness in responding.
Have a great day!!

pouriap, 2021-12-03:
Oh I'm happy it is working. But I suspect your problem is due to converting bytes to chars and vice versa. 
Your transmitter code didn't compile so I couldn't test it to be sure.
```
exit status 1
'ledL' was not declared in this scope
```

But there are a few things to consider:

1- There is no need to convert your byte to char. TinyRF works with binary data so you can just send your byte like this:
```c++
if (sendState == 1) {  //data is only send if a button was pressed
	ledStates = (ledStateL | ledStateR);
	digitalWrite(13, true); // Flash a light to show transmitting
	byte data[1] = {ledStates};
	sendMulti(data, 1, 3);
	delayMicroseconds(TX_DELAY_MICROS);
	digitalWrite(13, false);
}
```
This way you won't have to use the atoi() and itoa() functions and you also avoid possible overflows. 
An overflow can happen in this part of your code
```c++
for(int i=0; i<numRcvdBytes; i++){
	Serial.print((char)buf[i]);
	ledStatesChar[i] = ((char)buf[i]);  //save the data
}
```
Because you are not checking if numRcvdBytes is actually 1. If it's bigger than 1 then an overflow will occur because ledStatesChar is only 2 bytes long.

And in your RX code you can receive the byte like this:
```c++
void loop(){

	byte buf[1];
	uint8_t numLostMsgs = 0;
	uint8_t numRcvdBytes = 0;

	uint8_t err = getReceivedData(buf, 1, numRcvdBytes, numLostMsgs);

	if(err == TRF_ERR_SUCCESS){
		
		digitalWrite(13, true); // Flash a light to show received good message  

		Serial.print("Received: ");
		Serial.println(buf[0], BIN);
		
		ledStates = buf[0];
		
		digitalWrite(13, false);
		
		Serial.print("LED States: ");
		Serial.println(ledStates, BIN);
	}

	if (ledStates == 1){
		if ((unsigned long)(millis() - previousMillis) >= interval) {
			previousMillis = millis();
			colorWipe();
		}
	}
	else {
		strip.clear();
		strip.show();
		currentPixel = 0;
	}
	
	if(currentPixel == 15){
		strip.clear();
		strip.show();
	}
	
}
```

2- You mentioned you want to use an attiny as the receiver. I'm not sure if your code will fit in a tiny but you have to try and see. But one thing you should definitely do is reduce the RX buffer size. Go to `settings.h` and change

`#define TRF_RX_BUFFER_SIZE 128`

to 

`#define TRF_RX_BUFFER_SIZE 16`

16 bytes is big enough for you because you only send one byte. so even if you send it 5 times the code will still be able to buffer 3 messages: 1 bytes * 5 times * 3 messages = 15 bytes.

MicheQ, 2021-12-03:
Ok thanks for the suggestions, 
as soon as I try the code I will update you.

MicheQ, 2021-12-05:
I followed your advice. 

I also changed the code, because I read that it is complicated to send only zeros via RF, now to put all the LEDs off I do not send only zeros via RF but I send a "communication start" bit.

```
 ̶l̶e̶d̶S̶t̶a̶t̶e̶L̶ ̶=̶ ̶B̶0̶0̶;̶ ̶ ̶ ̶ ̶/̶/̶0̶0̶ ̶=̶ ̶f̶i̶r̶s̶t̶ ̶L̶E̶D̶ ̶o̶f̶f̶,̶ ̶0̶1̶ ̶=̶ ̶f̶i̶r̶s̶t̶ ̶L̶E̶D̶ ̶o̶n̶

ledStateL = B100;    //100 = first LED off, 101 = first LED on
```

Now the code works fine, I will definitely try to optimize it!

closed 2021-12-05 by MicheQ

pouriap, 2021-12-05:
Glad to hear it's working.
Also no need to worry about sending zeroes. TinfyRF's encoding is in such a way that both HIGH and LOW signals will be sent for each bit to avoid the problem you mentioned, so you can send as many zeroes as you want.

MicheQ, 2021-12-07:
For attiny85 is it right to connect the recevier on the PB2?
Also for transmitter?

pouriap, 2021-12-07:
Yes.
For transmitter it doesn't matter which pin you use.
For receiver it has to be a pin that has external interrupt. Which is PB2 on ATtiny85.