pouriap/TinyRF #6: Usage on Attiny85

yannik0707, 2022-01-20:
Hi,
have you ever tried using the Attiny85 as a reciever?
I'm trying to, but cant get it working. I already attached a 16 MHz crystal and tried different pin configurations and used the same code on a Atmega328p and it works. Is there anything left I can try to get it working?

pouriap, 2022-01-20:
I have not unfortunately. 

Do you get any errors?

What does your `Settings.h` file look like? Have you changed it?

yannik0707, 2022-01-20:
Since its hard to debug the Attiny85, I don't get any that could be usefull, but I used some LEDs to "display" the return of `getReceivedData()`, so its blinks, if an error accured. 
My `Settings.h´ is stock and I'm just sending some chars.
Also the Tiny is runnin, I'm adjusting a PWM Output with the transmitted data and the output works fine, but is not adjusting. 
I can also see the reciever is working, because there are bits coming in on the DATA pin when I'm transmitting data (looked close with a scope).
Is there any possibility that some INT-Pins are not supported or anything?

yannik0707, 2022-01-20:
*The error LED is not blinking, so I dont think there is no data coming through

pouriap, 2022-01-20:
You can only use PB2 because it has to be an external interrupt pin.

Try reducing your buffer size in `Settings.h` and see if that helps.
Change  `#define TRF_RX_BUFFER_SIZE 128` to something smaller like 16

The buffer size depends on how many bytes each message contains, how many times you send each message, and how quickly you call `getReceivedData()` in your code. For example if you want to send a 2 byte message 4 times (using sendMulti()), the buffer size should be at least (2+3)*4 = 20 bytes. (Those 3 extra bytes are message length, crc and sequence number added by the software).

yannik0707, 2022-01-20:
Thanks, I will try it and tell you

pouriap, 2022-01-22:
So I bought an ATtiny85 today! And can confirm that TinyRF does actually work on it as receiver. But you have to reduce the buffer size like I suggested. It doesn't work with the 128 byte buffer.

Other than the buffer size you have to be careful in your code to reduce your RAM usage as much as possible. That fat 30 byte buffer for instance in the receiver example is a no-no if you are using an ATtiny as receiver.

I'll update the examples and the code to add support and documentation for tiny85.

yannik0707, 2022-01-24:
Yeah it finally works, thank you very much

closed 2022-01-24 by yannik0707