pouriap/TinyRF #7: Receiver with ATtiny85 is not working with this code.

jigneshpatel175125, 2022-03-21:
Hi,
I'm trying the same receiver code with ATTiny85 but I'm not receiving anything, if I switch to Arduino UNO then it's working.
Can you let me know the possible solution?

pouriap, 2022-03-21:
Doesn't really matter, but higher is better because it till spend less time in the interrupt.

Remember to reduce your RAM usage throughout your code too. For example the 30-byte buffer in the receiver example is too big for attiny. You should reduce it to something smaller and send smaller messages.

pouriap, 2022-03-21:
Your data is an 8 character string so 9 bytes buffer should be enough.

Also that `data=buf` is not going to work as expected because you haven't specified the size of `data` so you cannot assign buf to it. You can use `buf` itself, no need to copy it to another variable. The receiver example shows how you can print buf to serial.

Also I'm assuming you reduced the buffer in `Settings.h` as I explained [here](https://github.com/pouriap/TinyRF/issues/6#issuecomment-1017562196).

jigneshpatel175125, 2022-03-21:
I tried by keeping 
 const uint8_t bufSize = 9 and #define TRF_RX_BUFFER_SIZE 16
 But still not able to receive anything. 
 Please see attached image.

[image]

pouriap, 2022-03-21:
Does this same code work with Arduino as receiver?

Have you tried the example receiver sketch with the new settings?

jigneshpatel175125, 2022-03-22:
Yes, the same code is working if I switch from ATtiny85 to Arduino Uno.
I'm using example sketch with new settings. 
[image]

pouriap, 2022-03-22:
Maybe the serial printing is taking too long and interfering with the interrupt.

What speed did you set your attiny CPU?

What is your serial baud rate?

Try the fastest CPU speed with a faster baud rate.

Also you should put your serial printing in an `if` statement like the receiver example to avoid unnecessarily printing empty messages:
```c++
  if(err == TRF_ERR_SUCCESS){
    Serial.print("Received: ");
    for(int i=0; i<numRcvdBytes; i++){
      Serial.print((char)buf[i]);
    }
    Serial.println("");
  }
```

Also please post your code as text not image.
Here's a howto: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

jigneshpatel175125, 2022-03-22:
I'm using ATtiny85 for receiver with #define TRF_RX_BUFFER_SIZE 16 and with clock speed of 8MHz
Please find my Tx and Rx code.

TX: 
#include "TinyRF_TX.h"

void setup()
{
  setupTransmitter();
}

void loop()
{
  const char* msg = "10000001";
  send((byte*)msg, strlen(msg));
  delayMicroseconds(TX_DELAY_MICROS);
}
********************
RX:
#include "TinyRF_RX.h"
#include "SoftwareSerial.h"

SoftwareSerial Monitor(5, 4);

uint8_t rxPin = PB2;

void setup()
{
Monitor.begin(9600); 
pinMode(4, OUTPUT);
pinMode(5, INPUT); 
Monitor.println("Setup Sucess");
}

void loop()
{
  const uint8_t bufSize = 9;
  byte buf[bufSize];
  uint8_t numLostMsgs = 0;
  uint8_t numRcvdBytes = 0;

  uint8_t err = getReceivedData(buf, bufSize, numRcvdBytes, numLostMsgs);
  if(err == TRF_ERR_SUCCESS)
  {
    Monitor.print("Received: ");
    for(int i=0; i<numRcvdBytes; i++){
      Monitor.print((char)buf[i]);
    }
    Monitor.println("");
  }
}

pouriap, 2022-03-22:
Is this your complete code?
I don't see a `setupReceiver(rxPin);` in your setup.

pouriap, 2022-03-22:
Also try increasing the serial baud rate to something like 115200 or 57600 or 38400.

pouriap, 2022-03-22:
I was actually going to update the repo and add examples and descriptions for using it with ATtiny85. Like I was planning to do it this week. So if you can wait a day or two I'm going to make that update.

pouriap, 2022-03-24:
I just updated the repo with official ATtiny85 support.
Download and install the `TinyRF_85-v1.7.zip` library from [here ](https://github.com/pouriap/TinyRF/releases/tag/v1.7).
A `TinyRF_85` should appear in your libraries. Make sure you use the **_85** version examples.

I am using ATTinyCore with these settings:

[image]

jigneshpatel175125, 2022-03-25:
Yes, thanks for the update!!
I'll try and update here.

jigneshpatel175125, 2022-03-25:
Just for refernce can you please provide your ATTiny85 board manager link ?

pouriap, 2022-03-25:
I'm not certain because I'm on vacation rn and don't have access to my PC. But I'm pretty sure it's this one:
https://github.com/SpenceKonde/ATTinyCore

Were you not able to run the example?

jigneshpatel175125, 2022-03-25:
Unfortunately No,
Were you able to see data on the serial monitor ?

pouriap, 2022-03-25:
Yes I was.
Have you tried the serial without TinyRF?

Post your wiring too.

jigneshpatel175125, 2022-03-25:
Don't forget proper powering! A 0.1uF decoupling cap for the MCU is mandatory. I personally recommend an additional 22uF across the MCU and at least 100uF across the transmitter module. Also use a nice and stable power source. This will minimize errors and headaches.

This was the problem with my ATTiny85, the proper powering!! My bad.

MANY MANY THANKS for helping me out!!!!
APPRECIATE !!!!!!!!!!!!

pouriap, 2022-03-26:
No problem.

At least you made me release the tiny85 update I've been meaning to do for so long.

BTW I noticed you are sending a number as string in your code. I have included an example in the new release that shows how you can send numbers more eficiently.

closed 2022-03-26 by pouriap

jigneshpatel175125, 2022-03-26:
Hi @pouriap,

Thank you again for helping me out.

Yes, I tried your sending number code and it's working perfectly fine for me.

This for all people who will visit this thread in future.
PLEASE MAKE SURE TO CONNECT 0.1uF decap ACROSS MCU (Attiny85/Attiny13 or any standalone IC). Because author already mentioned to connect it but there are people like me who not read full description. 
Because if you tried on any development board it will work seemlessly and the reason is there are decap already mounted on boards.

Attaching image for reference.
[image]