pouriap/TinyRF #8: TinyRf for ATtiny85 with 1Mhz

arduino4me, 2022-04-22:
I used the TinyRF on an ESP8266 with the example of the library "Receiver" as an receiver.
As Transmitter I want to use a ATtiny85 with a battery.
To test the communication between ATtiny85 (transmitter) and ESP8266 as receiver I use on the ATtin85 your example "Transmitter" from the  TinyRF_85-library.
The test was ok. The ESP8266 shows "Received: Hello!". The bootloader was burned with 8Mhz in the Arduino-IDE.
Next step I want the ATtiny85 with 1Mhz (lower energy for battery).
So I burned the bootloader with 1Mhz and I used the same sketch as with 8Mhz.
Now there was no output on the ESP8266.
What can I do to use the ATtiny with 1 Mhz.

Thanks for support!

pouriap, 2022-04-23:
Have you tried using a lower data rate?

Go to `Settings.h` and in [line 62](https://github.com/pouriap/TinyRF/blob/8350d5192cacd21ccf163e9bad43428d8dd8a857/src/Settings.h#L62) change the data rate to `TRF_BITRATE_500` or `TRF_BITRATE_200`.

If I recall correctly I have already tried with 1Mhz oscillator on ATtiny13 and it worked with lower data rates.

BTW TinyRF_85 is only needed if you are using an ATtiny as receiver. For an ATtiny transmitter the normal library works fine.

arduino4me, 2022-04-23:
Thank you for this hint!
I used #define TRF_BITRATE_200, and for both sketches (ESP8266 and ATtiny85) the same library ATtinyRF (not the ATtinyRF_85).
It works with ATtiny85 and 1 Mhz!
A great library for an environment with low memory, like ATtinys!

pouriap, 2022-04-23:
Glad to hear that. 👍

closed 2022-04-23 by pouriap