pouriap/TinyRF #11: TinyRF85 example does not compile

themelispan, 2023-06-30:
I tried compiling the TinyRF85 Receiver example with board "Digispark (Default - 16.5MHz)" which is my digispark ATtiny85 board. I chaged the line #include <SoftwareSerial.h> to #include <SoftSerial.h> because the compiler thought it should use the ESP8266/ESP32 SoftwareSerial library. Still, I am getting the error 'SoftwareSerial' does not name a type.

C:\Users\theme_9p6cezr\OneDrive\Arduino\Tiny85_RC_car_receiver\Tiny85_RC_car_receiver.ino:13:1: error: 'SoftwareSerial' does not name a type
 SoftwareSerial mySerial(3, 4);  //rx, tx
 ^
C:\Users\theme_9p6cezr\OneDrive\Arduino\Tiny85_RC_car_receiver\Tiny85_RC_car_receiver.ino: In function 'void setup()':
C:\Users\theme_9p6cezr\OneDrive\Arduino\Tiny85_RC_car_receiver\Tiny85_RC_car_receiver.ino:31:3: error: 'mySerial' was not declared in this scope
   mySerial.begin(115200);
   ^
C:\Users\theme_9p6cezr\OneDrive\Arduino\Tiny85_RC_car_receiver\Tiny85_RC_car_receiver.ino: In function 'void loop()':
C:\Users\theme_9p6cezr\OneDrive\Arduino\Tiny85_RC_car_receiver\Tiny85_RC_car_receiver.ino:54:5: error: 'mySerial' was not declared in this scope
     mySerial.println("Buffer too small for received data!");
     ^
C:\Users\theme_9p6cezr\OneDrive\Arduino\Tiny85_RC_car_receiver\Tiny85_RC_car_receiver.ino:59:5: error: 'mySerial' was not declared in this scope
     mySerial.println("Received corrupted data.");
     ^
C:\Users\theme_9p6cezr\OneDrive\Arduino\Tiny85_RC_car_receiver\Tiny85_RC_car_receiver.ino:66:5: error: 'mySerial' was not declared in this scope
     mySerial.print("Received: ");
     ^

exit status 1

Compilation error: 'SoftwareSerial' does not name a type

pouriap, 2023-07-01:
I haven't used that library but I guess you should change the line
```c++
SoftwareSerial mySerial(3,4);  //rx, tx
```
to
```c++
SoftSerial mySerial(3,4);  //rx, tx
```

themelispan, 2023-07-05:
I tried it, it doesn't work. It will only compile if I comment out all serial-related commands

pouriap, 2023-07-05:
Does the library have an example sketch? Does the example sketch compile?